neonious/lowjs #120: Can not use Socket from net module.

iamcco, 2020-09-19:
I try lowjs and have issue with socket.

index.js

``` javascript
const net = require('net')

const client = net.Socket()
```

run the code `./bin/low index.js`

> same error when run on esp32 board.

error:

```
TypeError: undefined not callable (property 'on' of [object Object])
    at [anon] (deps/duktape/src-low/duktape.c:67641) internal
    at Readable (lib:stream:48) strict
    at Duplex (lib:stream:500) strict
    at Socket (lib:net:124) strict
    at [anon] (/Users/aioiyuuko/my/watching/index.js:3) preventsyield
```

ThomasRogg, 2020-09-19:
Hi!
can you try net.createServer or net.connect instead of net.Socket() to create the socket, depending on whether you want to create a server or connect to a server, and then report again?
Thank you!
Thomas

iamcco, 2020-09-21:
I test with connect and it works fine, thanks!

closed 2020-09-21 by iamcco