neonious/lowjs #123: make error on build

GiovanniCardamone, 2020-12-27:
Hello, i was trying to build a low.js bin on my machine

```
$ uname -a
Linux Jhos 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

but the make command gave me the following error:

```
$ make         
rm -rf deps/duktape/src-low
cd deps/duktape && python tools/configure.py --output-directory=src-low	\
	-DDUK_USE_GLOBAL_BUILTIN   \
	-DDUK_USE_BOOLEAN_BUILTIN   \
	-DDUK_USE_ARRAY_BUILTIN   \
	-DDUK_USE_OBJECT_BUILTIN   \
	-DDUK_USE_FUNCTION_BUILTIN   \
	-DDUK_USE_STRING_BUILTIN   \
	-DDUK_USE_NUMBER_BUILTIN   \
	-DDUK_USE_DATE_BUILTIN   \
	-DDUK_USE_REGEXP_SUPPORT   \
	-DDUK_USE_MATH_BUILTIN   \
	-DDUK_USE_JSON_BUILTIN   \
	-DDUK_USE_BUFFEROBJECT_SUPPORT   \
	-DDUK_USE_ENCODING_BUILTINS   \
	-DDUK_USE_PERFORMANCE_BUILTIN   \
	-DDUK_USE_OBJECT_BUILTIN    \
	-DDUK_USE_ES6_PROXY	\
	-DDUK_USE_GLOBAL_BINDING \
	-DDUK_USE_SYMBOL_BUILTIN	\
	-DDUK_USE_SECTION_B \
	-DDUK_USE_CPP_EXCEPTIONS
Traceback (most recent call last):
  File "tools/configure.py", line 1005, in <module>
    main()
  File "tools/configure.py", line 315, in main
    get_duk_version(os.path.join(srcdir, 'duktape.h.in'))
  File "tools/configure.py", line 139, in get_duk_version
    m = r.match(line)
TypeError: cannot use a string pattern on a bytes-like object
make: *** [Makefile:98: deps/duktape/src-low/duktape.c] Errore 1
```

someone know what the problem is?

ThomasRogg, 2020-12-27:
My guess: python most probably links to v3, but you must have v2 installed for DukTape.

GiovanniCardamone, 2021-01-09:
yep, my python executable links to python3.7

```bash
$python --version
Python 3.7.7
```

closed 2021-01-10 by GiovanniCardamone