neonious/lowjs #124: MQTT pub issue

seblinux-enod, 2021-01-05:
Hi,

I try to perform a simply MQTT test with lowjs with a roover esp32 8 MB.

I can connect, subscribe to a mqtt broker. 
I can see in the monitor all payload publish with an other computer. 
I code a function to publish any 5 sec on the same topic but I canot see anything. No error also.

somone knows this issue ?

Br

ThomasRogg, 2021-01-05:
This question is far to general to answer.

First, I would try the same code with Node.JS on a PC. If that does not work, the problem has nothing to do with low.js. If it works, it might be low.js which is doing something wrong. Then I would step through the code, maybe with console.logs and figure out where exactly things behave differently.

seblinux-enod, 2021-01-06:
Hi,

The code is working with node.js on a pc, no problem to publish in local mqtt broker...

Br

seblinux-enod, 2021-01-06:
Hi,

for the publish I have to do a : client.connect before the publish.

so I think that there are a probleme with the mqtt connection or with the main loop but for my side I think it is not the mail loop because I put a counter and it is increase when I do the pub...

Br

ThomasRogg, 2021-01-06:
So you have a workaround with the client.connect?

Interesting to read your story, but I cannot really help without more input.

paolodinoto, 2021-02-19:
I had a similar issue:
- board connect to broker, subscribe to some topics and publish on others immediately after connection, everythings works fine
- when an update occurs, board try to publish the update, no code error, but no message is sent.

After a fast seach into MQTT.js code, i have enabled debug messages and i have found that, for some reason (i have not investigated), the messages are put in a queue/buffer instead of being sent.

As a fast workaround i have added this line of code before a publish call (disabling the use of queue/buffer):

`
// mqtt is the client instance returned after a connect
mqtt._storeProcessing = false;
mqtt.publish(topic, value, {qos: 1, retain: true});
`

ThomasRogg, 2021-02-21:
You might want to try the new version 1.6.1, flashable now on ESP32. The PC version is not available yet, will be in a day or so.

A fix I did might help you here. Or might not, I do not know.