neonious/lowjs #126: SyntaxError: unterminated statement (line 7)

atmoner, 2021-01-27:
Hello,
The first compilation is going correctly, the example works but when I send a new scetch, I constantly have this error:

[image]

and the client becomes unreachable :/

ThomasRogg, 2021-01-28:
This is because you turned transplation off. low.js only supports ES5.1. So keywords like let do not work.

I just deployed lowsync 1.4.10. Can you check whether that version works with transpilation but without the invariant not found message? Might, but might not, as I am tapping in the dark: I cannot reproduce your problem on my own computer (Mac)..

Thomas

atmoner, 2021-01-28:
> This is because you turned transplation off. low.js only supports ES5.1. So keywords like let do not work.
>
> I just deployed lowsync 1.4.10. Can you check whether that version works with [...]

Hello Thomas,
indeed, I had to adapt to understand the problem by deactivating the transplation...
I updated lowsync to version 1.4.10, I activated the transplation, I uploaded a new js code and finally, it works great :+1: 

Thank you very much for fixing my problem.

closed 2021-02-21 by ThomasRogg