neonious/lowjs #127: M5Stack button pin not available

jakpaw, 2021-02-26:
Hi, thanks for the great software, I really like the idea of using JavaScript to program microcontrollers :) I have a minor problem with using lowjs on M5Stack Fire device. It has 3 buttons, attached to pins 37, 38 and 39: https://forum.m5stack.com/topic/360/m5stack-fire-pinout-leaflet. I can add an event listener to pin 39 like this `gpio.pins[39].on("fall", () => {})` and it works, but doing the same for pins 38 or 37 gives an error: `RangeError: pin 38 is not available on your device`. Could you look into it? I'm a beginner in the world of microcontrollers but from what I could find on the internet, these pins are somehow special: https://github.com/espressif/arduino-esp32/issues/397

ThomasRogg, 2021-02-28:
Pin 37 and 38 were disabled to use because ESP32-WROVER does not have these pins. However, the board you have has ESP32 and PSRAM seperate and thus has pin 37 and 38.

Reenabled 37 and 38 in new version 1.6.2. Can be flashed with lowsync now.

-- Thomas

jakpaw, 2021-03-01:
Perfect, thanks!

closed 2021-03-01 by jakpaw